4-20ma/ModbusMaster #13: basic example.. error compiling

Ilyassss, 2015-05-21:
Arduino: 1.6.1 (Windows 8.1), Board: "Arduino Uno"

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:223:0,

                 from C:\Users\Ilyas\Documents\Arduino\libraries\ModbusMaster-master\ModbusMaster.h:53,

                 from C:\Users\Ilyas\Documents\Arduino\libraries\ModbusMaster-master\ModbusMaster.cpp:32:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial.h: In member function 'HardwareSerial& HardwareSerial::operator=(const HardwareSerial&)':

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial.h:83:7: error: non-static const member 'volatile uint8_t* const HardwareSerial::_ubrrh', can't use default assignment operator

 class HardwareSerial : public Stream

       ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial.h:83:7: error: non-static const member 'volatile uint8_t* const HardwareSerial::_ubrrl', can't use default assignment operator

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial.h:83:7: error: non-static const member 'volatile uint8_t* const HardwareSerial::_ucsra', can't use default assignment operator

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial.h:83:7: error: non-static const member 'volatile uint8_t* const HardwareSerial::_ucsrb', can't use default assignment operator

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial.h:83:7: error: non-static const member 'volatile uint8_t* const HardwareSerial::_ucsrc', can't use default assignment operator

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial.h:83:7: error: non-static const member 'volatile uint8_t* const HardwareSerial::_udr', can't use default assignment operator

C:\Users\Ilyas\Documents\Arduino\libraries\ModbusMaster-master\ModbusMaster.cpp: In member function 'void ModbusMaster::begin(uint16_t)':

C:\Users\Ilyas\Documents\Arduino\libraries\ModbusMaster-master\ModbusMaster.cpp:139:16: note: synthesized method 'HardwareSerial& HardwareSerial::operator=(const HardwareSerial&)' first required here 

       MBSerial = Serial;

                ^

Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

4-20ma, 2015-05-21:
What version of the ModbusMaster library are you using (look in the library.properties file)? This looks like a fork of an old version.

4-20ma, 2016-09-10:
Closing issue due to the age.

closed 2016-09-10 by 4-20ma