4-20ma/ModbusMaster #30: 0.11.0 very slow writeMultipleRegisters

TheGrovesy, 2015-05-27:
This might also be true for the other Modbus Functions, however I am currently only using writeMultipleRegisters.
With version 0.10.3 (and also 0.10.2 & 0.9.1), using the milli() function, the time it takes to perform a node.setTransmitBuffer, followed by a node.writeMultipleRegisters is around 209 milliseconds. If I use exactly the same sketch with version 0.11.0 the difference is almost exactly 10 times slower at 2006 milliseconds.

Here is my code to check this:

void loop() {
  static uint32_t i = 0;
  static long previousMillis = 0;
  unsigned long currentMillis = 0;
  uint8_t result;
  
  // put your main code here, to run repeatedly:
  previousMillis = millis();
  node.setTransmitBuffer(0, lowWord(i++));
  result = node.writeMultipleRegisters(3960, 1);
  currentMillis = millis();
  
  Serial.println(currentMillis - previousMillis);
  
  //delay(1000);
}

4-20ma, 2016-09-10:
Looks like it might be the following addition between v0.10.3 and v0.11.0:

``` c++
while (MBSerial->read() != -1);
```

You might try commenting this line out to see if it makes a difference.

4-20ma, 2016-09-10:
Closing issue. Re-open or create a PR if you have a proposed solution. Thanks.

closed 2016-09-10 by 4-20ma