4-20ma/ModbusMaster #31: Minor fix for the ESP8266 compiler

rpseng, 2015-05-31:
Hi, this is a really mice library!

After some work I got it working within the ESP8266 wifi module together with a max3232 chip. This is less than USD 5 to get a RTU Modbus hardware by wifi!

The only problem is that the esp compiler gets confused with your ModbusMaster::send function. The fix is very simple, just adjust the cast:
void ModbusMaster::send(uint8_t data)
{
  send((word)data);
}

We also discussed this issue here:
https://github.com/esp8266/Arduino/issues/360

All the best.

aneek231, 2016-05-30:
If this library works with ESP8266? I get this during compilation: This library only supports boards with an AVR or SAM processor. Please open an issue at https://github.com/4-20ma/ModbusMaster/issues and indicate which processor/platform you're using."

4-20ma, 2016-09-10:
As of #29, you should be able to compile against the ESP8266.

closed 2016-09-10 by 4-20ma