4-20ma/ModbusMaster #32: Code not work with arduino mega 2560 115200 baud rate

tuhuynh, 2015-06-16:
I used this code to read holding register with baud rate  setup is 115200. it didn't work. If i set 57600 baud rate it work very good. I don't know this error from my Mega 2560 or modbus master code???

tuhuynh, 2015-06-18:
If after "node.begin(115200)" i use "Serial.begin(115200)", code modbus work good. I think this modbus lib have a broblem but don't find out what happend make this error with mega 2560.

tuhuynh, 2015-06-19:
Function readHoldingRegister() use to read up to 125 16bit registers, but responebuffer array have only 64 16bit element so it can't fit for readHoldingRegister() function when read 125 register at once. We must increase ku8MaxBufferSize to 125 to fix this problem.

4-20ma, 2016-09-10:
Can you confirm that this works after changing `ku8MaxBufferSize` to 125?

4-20ma, 2016-09-23:
Closed due to inactivity.

closed 2016-09-23 by 4-20ma