4-20ma/ModbusMaster #33: How is RS485 Tx Enable controlled

keithfisk, 2015-06-17:
I am fairly new to Arduinos and am looking for an easy to use Modbus RTU master.  With your ModbusMaster how is the Tx enable for the RS485 driver controlled?  I can't see anywhere in your library where this can be set up. I expected the ModbusMaster.begin to allow the TxEn signal to be allocated to any Arduino digital out pin. All I can see for .begin are parameters for choosing the Arduino serial port and the Modbus Id for the slave.

arifonder, 2015-09-11:
I'm also wondering this.

ekarinca, 2015-11-24:
With Example Program
I added **pinMode(3, OUTPUT);** where MAX485 Enable pin

```javascript
#include <ModbusMaster.h>
// instantiate ModbusMaster object as slave ID 2
// defaults to serial port 0 since no port was specified
ModbusMaster node(2,1);
void setup()
{
  pinMode(3, OUTPUT);
  // initialize Modbus communication baud rate
  node.begin(9600);
  Serial.begin(9600);
}
void loop()
{
  static uint32_t i;
  uint8_t j, result;
  uint16_t data[6];
  i++;
  // set word 0 of TX buffer to least-significant word of counter (bits 15..0)
  node.setTransmitBuffer(0, lowWord(i));
  // set word 1 of TX buffer to most-significant word of counter (bits 31..16)
  node.setTransmitBuffer(1, highWord(i));
  // slave: write TX buffer to (2) 16-bit registers starting at register 0
  //result = node.writeMultipleRegisters(0, 2);
  // slave: read (6) 16-bit registers starting at register 2 to RX buffer
  result = node.readInputRegisters(105, 6);
  // do something with data if read is successful
  if (result == node.ku8MBSuccess)
  {
    for (j = 0; j < 6; j++)
    {
      data[j] = node.getResponseBuffer(j);
      Serial.print(j); Serial.print("\t->\t"); Serial.println(data[j]);
      
    }
  }
  Serial.print("\n");
  delay(1000);
}
```

And on ModbusMaster.cpp library somewhere at 760th line

I added **digitalWrite(3, HIGH);** and 
**digitalWrite(3, LOW);** lines

I am still testing but for now its working

```javascript
  // flush receive buffer before transmitting request
  while (MBSerial->read() != -1);

  // transmit request
  digitalWrite(3, HIGH);        //RS485 pin
  for (i = 0; i < u8ModbusADUSize; i++)
  {
#if defined(ARDUINO) && ARDUINO >= 100
    MBSerial->write(u8ModbusADU[i]);
#else
    MBSerial->print(u8ModbusADU[i], BYTE);
#endif
  }
  
  u8ModbusADUSize = 0;
  MBSerial->flush();    // flush transmit buffer
  
  digitalWrite(3, LOW);        //RS485 pin
  // loop until we run out of time or bytes, or an error occurs
```

kintel, 2016-09-08:
This is important for half-duplex setups. I've tested the solution suggested by @ekarinca and I think we should add support for this. I see two main options:
1. Add configurable pin numbers for RS385 Driver Enable and Receiver Enable (usually these can be connected to the same pin, but sometimes you'd want them separate).
2. Add a callback just before transmitting and just after the transmission is done. Since there is already an idle callback in the API, this sounds like a natural extension.

@4-20ma If there is interest in merging smth. like this I might be able to provide a pull request.

commit 189b8ba9 (2016-09-11): Added preTransmission() and postTransmission(). Fixes #33
commit f22723fd (2016-09-11): Merge pull request #44 from kintel/issue33

Added preTransmission() and postTransmission(). Fixes #33
closed 2016-09-11 by 4-20ma