4-20ma/ModbusMaster #34: Use the library with Arduino Uno

bojankoce, 2015-07-01:
Hi buddy (4-20mA),

I just found that your github ModbusMaster library perfectly match my needs. Namely, I have an ElNet LT Energy&Powermeter (http://www.getelnet.com/wordpress/2009/10/15/elnet-lt/) that supports Modbus RTU communication an has built-in Functions 03, 04, 06, and 16.
I'm also planning to use max485 ESA transceiver and Arduino Uno board.

Now, I would like to know is it possible to use your library with Arduino Uno board? I saw that you made testings by using Duemilanove board that is similar to Uno bot not the same.
Also, I'm confused how to connect max485 pins:
- For RO and DI pins, I know that they are connected to Arduino serial pins (0 and 1, respectively)
- For A and B pins, I know that they are connected with the ElNet LT Powermeter device
- For Vcc pin I know it is connected to 5V DC

- What I don't know is where to connect on Arduino board RE and DE pins of max485. Can you help me with this?

Thank you very much for your time and effort to help me.

Sincerely,
Bojan.

vivekkaundal, 2016-05-11:
Thr library worked fine with duemilanove. I hvnt check it with arduino uno. And short pin RE and DE and connect it with any digital pin of arduino. Bt configure that digital pin to set directionality of max 485 as max 485 is a half duplex

4-20ma, 2016-09-10:
Did you get your question answered?

4-20ma, 2016-09-23:
Closed due to inactivity.

closed 2016-09-23 by 4-20ma