4-20ma/ModbusMaster #38: Arduino DUE and Serial1,2,3

silas15, 2016-03-30:
Hi, thanks for super library.

I have problem with using different serial other then serial 0. On UNO si code fast and without problems. On DUE works slowly (but works). Serial1(2,3) on DUE didn't work at all. It sends nothing. I tried:
`ModbusMaster node(Serial1,1);`
`ModbusMaster node(1,1); `
Comunication not working.

I using library version 0.11.0, arduino IDE 1.6.8

Only corrections I made, is adding lines for TX enable and
`MBSerial->begin(u16BaudRate, SERIAL_8E1)` 

Code works on UNO without any problems.

theurlm, 2016-06-08:
Hy, can you show me the code of the modifications of TX enable and SERIAL_8E1 ?

silas15, 2016-06-11:
I found the problem. Below line:
`MBSerial->begin(u16BaudRate);`
i add
`Serial1.begin(9600, SERIAL_8E1);`
Then serial prot works with parity.
I don't understand how it works, but i'm happy.

For TX enable I just add digital write high function above line:
`MBSerial->flush();`

4-20ma, 2016-09-10:
Glad it's working. Thanks. Closing issue.

closed 2016-09-10 by 4-20ma