4-20ma/ModbusMaster #45: Change license to Apache 2.0

4-20ma, 2016-09-10:


kintel, 2016-09-11:
For the record, feel free to relicense any contributions I've made under any Open Source license.

4-20ma, 2016-09-11:
When I originally wrote ModbusMaster, it was partly to solve a problem (extract real-time data from a PLC), and partly as an exercise in learning. I was inspired by and borrowed concepts from [Arduino Modbus Master](http://sites.google.com/site/jpmzometa/arduino-mbrt/arduino-modbus-master). I believe in Open Source Software and the concept that sharing/applying ideas from multiple contributors make a better overall product.

This library is successful when people use it to solve real-world problems and contribute their improvements to the project. It may be used in open source, public domain, private, or commercial projects. When someone links to this library, it becomes a derivative work of this code. The GPLv3 requires distribution of this combined work to be covered under GPLv3 or other compatible license. The Apache 2.0 license is more appropriate for this project because it is more open with respect to distribution of derivative works, as compared to GPLv3. Therefore, as of v0.12.0, I'm changing the project's license from GPLv3 to the less restrictive Apache 2.0 license.

PR #67 (2016-09-11): 45 change license to apache 20

### Description
Changes license from GPLv3 to Apache 2.0

### Issues Resolved
Resolves #45.

### Check List

General

- [x] Code follows coding style defined in STYLE.md
- [ ] Doxygen comments are included inline with code
- [x] No unnecessary whitespace; check with `git diff --check` before committing.

The following have been modified to reflect new features, if warranted

- [x] README.md
- [ ] keywords.txt (use tabs as whitespace separators)
- [ ] library.properties
- [x] examples/ - update or create new ones, as warranted

The following have **NOT** been mod [... cut]
commit d591ac31 (2016-09-11): Resolves #45 Change license to Apache 2.0
closed 2016-09-11 by 4-20ma