4-20ma/ModbusMaster #77: EM-6433 register, Modbus, uno

vindhyachaltakniki, 2016-09-17:
1. I ma using this modbus library to read (EM)6433, meter from scheider.
Lib: https://github.com/4-20ma/Modbus...
doc: http://4-20ma.io/ModbusMaster/
 
2. (EM)6433 documentation is here: http://tinyurl.com/z7696ys 
 
3. Page 3 of above doc says that protocol is Modbus RTU, data:8 bits, baud: 9600(default) , parity:even, device address:1, stop bit:1

In uno code, default seting is 9600 & parity zero. Is below setting ok in code?
    ModbusMaster node;
    Serial.begin(9600, SERIAL_8E1);
    node.begin(1, Serial);

4. (EM)6433 doc on page 7 shows, read register "Average current", with address "3913" & return type is float 4 bytes in little endian.
However this library has function "readHoldingRegisters" to read holding register. But this function reads only 16 bit registers. 
So should I use the function as:

result = node.readHoldingRegisters(3913, 2);
if (result == node.ku8MBSuccess)
{
   for (j = 0; j < 2; j++)
   {
       data[j] = node.getResponseBuffer(j);
   }
}
parameter 2, shows that 2 16 bit values are read.
How to convert it back to float?

4-20ma, 2016-09-17:
1. OK
2. OK
3. This looks correct; you'll initialize the Serial port using the standard Arduino syntax specifying baud rate, size, parity, and stop bit(s). Device is address #1 on the RS485 network.
4. Modbus holding registers are 16-bit. Page 57 of the manual referenced in #2 above indicates "In the transfer of individual data values, it treats two registers as an object with the starting address (e.g., 3900) considered as the object name." Your code looks correct to read 2 consecutive registers starting at 3913 for the current average. You'll need to assemble these unsigned 16-bit values into an unsigned 32-bit value and convert to float. This [spreadsheet](http://www.simplymodbus.ca/ieeefloats.xls) appears to do what you desire.

vindhyachaltakniki, 2016-10-05:
1. I have interfaced EM-6433 using arduino over Rs485. Rs485 IC used is SN75HVD08. 

2. EM-6433 settings are slave id:1 & baud:9600, even parity & 1 stop
Confirmed it aftr turing on meter & go to its settings table.

3. Case 1: If I permanently pull down N_RE pin & pull up DE pin of SN75HVD08, then when I used the attached code I always receive 0xE2 in return from device ie ku8MBResponseTimedOut. In code ignore the code which ouput high & low the RE & DE pin. This is used in case 2. In this case, this is commented out.

4. Case2: Since this SN75HVD08 is half duplex so I connected RE & DE pins to arduino as in attached circuit. Also I have modified ModbusMaster.cpp file in line 728 & 729, where I added code to disable Tx & enable rx from IC. When I use this code I receive error 0x02 in result, which is ku8MBIllegalDataAddress.
What am I doing wrong here. I also interchanged A & B pins to slave device EM-6433 + & -. But no correct result obtained.

5.All ckt diagram and code is here: https://drive.google.com/open?id=0B-jbeBk3wfkWaWo1ek5ISUxoTzA

4-20ma, 2016-10-14:
I don't see anything obvious, but I'm not familiar with your device. Here's how I would proceed: start with as small a working example as possible and build upon that.

1. Get one of the examples working with your device. Remove any extraneous code and start from as basic a setup as possible. This ensures your baud rate, parity, stop are correctly applied.
2. Build upon the example by adding layers of complexity (LCD display, etc).

Do you use a logic analyzer such as a [Saleae](https://www.saleae.com) Logic or similar unit? This will help you decode the signal to ensure it's being sent/received correctly. Note that I don't have any relationship with Saleae, other than as a satisfied customer.

vindhyachaltakniki, 2016-10-15:
Hi
1. SN75HVD08 has Rx & Tx enable pins. I have shorted them, & in the code I keep this pin high, which means keep on transmitting any data.
In the code snippet below I added a line which brings pins low. This means now receive. Below is code which i have edited in your library:

  if (_postTransmission)
  {
    _postTransmission();
  }
  
  digitalWrite(2, LOW);
  
  // loop until we run out of time or bytes, or an error occurs
  u32StartTime = millis();
  while (u8BytesLeft && !u8MBStatus)
  {
    if (_serial->available())
    {
#if __MODBUSMASTER_DEBUG__
      digitalWrite(__MODBUSMASTER_DEBUG_PIN_A__, true);
#endif
      u8ModbusADU[u8ModbusADUSize++] = _serial->read();
      u8BytesLeft--;
#if __MODBUSMASTER_DEBUG__
      digitalWrite(__MODBUSMASTER_DEBUG_PIN_A__, false);
#endif
    }
    

2. As soon as receive is done again bring the pin to high, this way I am able to get correct data.
Is it ok?

3. Second, I am reading mutliple slave devices in a line. How to send each slave id. Do I have to evrytime initialize like this:

    node.begin(slave_id, Serial1);
    delay(100);    
    digitalWrite(r_enable, HIGH);   /* rx enable */
    result = node.readHoldingRegisters(address - 1U , 2U);
    digitalWrite(r_enable, HIGH);   /* rx enable */

4-20ma, 2016-11-19:
To read multiple devices, you'll instantiate each one as a separate object:

``` cpp
...
ModbusMaster device1;
ModbusMaster device2;
...
Serial.begin(19200);
device1.begin(1, Serial);
device2.begin(2, Serial);
...
```

sachinsinghal9, 2016-12-13:
Hi VindhyaChaltakniki,

I am also working on the same project, that is to query a EM6400 multifunction meter for different parameters using an arduino module, i wanted to know if you got the results properly. I am using the same library that you are using, however i am using Max485 ic as a RS485 to TTL Converter, and in Max485 it automatically switches between Transmission and Recieve Modes. So you may use that for it if you are still stuck with it, I wanted to know is your slave working with a Software such as MODSCAN as a master?? My EM6400 is responding well with this software but with the arduino , it is giving an exception of timeout, what can be the problem?

santhoshEsp32, 2017-04-09:
Hello ,
i'm trying to read veritech MFM using this library, but im getting response 0xF2 that timeout . i tried using halfdyplex example still im not finding output. and i wanted to read input registers of MFM.
im using software serial instead of hardware is that the problem ?    

here is my code 
void loop()
{
  static uint32_t i;
  uint8_t j, result;
  uint16_t data[6];
  
  result = node.readInputRegisters(53, 2);
  Serial.println(result);
  // do something with data if read is successful
  if (result == node.ku8MBSuccess)
  {
    for (j = 0; j < 6; j++)
    {
      data[j] = node.getResponseBuffer(j);
    }
  
  Serial.print(data[0]);
  Serial.print(data[1]);
  Serial.print(data[2]);
  Serial.println(data[3]);
  delay(1000);
  }
}

can please explain two parameters of this function - result = node.readInputRegisters(53, 2); 
1st parm i understood that its add of reg what about second parm? is that the pointer to store output?

thank you

gau999, 2017-06-06:
Hello sir, 
i'm also trying with EM6436 energy meter from Schneider

can you please provide me the code which you have written for em6433

slowpoison4, 2017-12-15:
@sachinsinghal9 Hi! Doing A similar project with EM1200. Were you able to find a solution? CAn you share me the details?

kprakash811, 2018-09-04:
@gau999 did u find the solution

sthiruppathi, 2020-01-12:
Hello All, did you find the solution?
could you please share the ideas?
by
thiruppathi
sthiruppathiece@gmail.com

sthiruppathi, 2020-01-18:
@gau999 @kprakash811  did u find the solution?