4-20ma/ModbusMaster #82: Update Particle library to latest version

vshymanskyy, 2017-01-16:
### ModbusMaster version
[Version of the project where you are encountering the issue]

### Arduino IDE version
[Version of Arduino IDE in your environment]

### Arduino Hardware
[Hardware information, including board and processor]

### Platform Details
[Operating system distribution and release version]

---

### Scenario:
[What you are trying to achieve and you can't?]

### Steps to Reproduce:
[If you are filing an issue what are the things we need to do in order to repro your problem? How are you using this project or any resources it includes?]

### Expected Result:
[What are you expecting to happen as the consequence of above reproduction steps?]

### Actual Result:
[What actually happens after the reproduction steps? Include the error output or a link to a gist if possible.]

---

### Feature Request

#### Narrative:

```` text
As a [role]
I want [feature]
So that [benefit]
````

#### Acceptance Criteria:

```` text
Scenario 1: Title
Given [context]
  And [some more context]...
When  [event]
Then  [outcome]
  And [another outcome]...
````

4-20ma, 2017-02-07:
The issue request form is designed to be filled out so questions between [brackets] indicate the nature of the request. From the title, it appears you'd like something updated for particle.io. Please contact the author of the particle.io library--it appears it's a fork off my ModbusMaster library, but I don't maintain their fork.

closed 2017-02-07 by 4-20ma

vshymanskyy, 2017-02-07:
Thanks. Sorry, didn't notice that!