4-20ma/ModbusMaster #86: ESP8266 Crash when RS485 is disconnected from MAX485

luisgcu, 2017-04-10:
### ModbusMaster version
[2.0.1]

### Arduino IDE version
[1.8.2]

### Arduino Hardware
[ESP8266, NodeMCU1.0 ESP-12E]

### Platform Details
[Operating system distribution and release version]

---

### Scenario:
[Read  holding register (10) from a VFD, Max485 is conected to GPIO, 14,12 on the ESP8266 using the software serial]

### Steps to Reproduce:
[II am able to read modbus registers without problem , the only problem that I have  is, if I disconnect the  RS485 conection from the MAX485 then the ESP8266 crash and reset  over and over again]

###THE CODE ###
#include <ModbusMaster.h>
#include <SoftwareSerial.h>

/*!
  We're using a MAX485-compatible RS485 Transceiver.
  Rx/Tx is hooked up to the hardware serial port at 'Serial'.
  The Data Enable and Receiver Enable pins are hooked up as follows:
*/
#define MAX485_DE     5 // was  5(moteino)  //and was 2(d4) on esp8266 but have trouble
#define MAX485_RE_NEG  4 // was 6
#define NANO_HSP(m,n)  (m *100+ n-1) ///<   

ModbusMaster node;

SoftwareSerial mySerial(14, 12, false,256); //esp8266
void preTransmission()
{
  digitalWrite(MAX485_RE_NEG, 1);
  digitalWrite(MAX485_DE, 1);
}

void postTransmission()
{
  digitalWrite(MAX485_RE_NEG, 0);
  digitalWrite(MAX485_DE, 0);
}

void setup()
{
  pinMode(MAX485_RE_NEG, OUTPUT);
  pinMode(MAX485_DE, OUTPUT);
  
  // Init in receive mode
  digitalWrite(MAX485_RE_NEG, 0);
  digitalWrite(MAX485_DE, 0);
Serial.begin(19200);
  // Commander SK @ 19200
  mySerial.begin(19200);
Serial.println("Modbus-Master");

  // Modbus slave ID 1
  node.begin(1, mySerial);
  // Callbacks allow us to configure the RS485 transceiver correctly
  node.preTransmission(preTransmission);
  node.postTransmission(postTransmission);
}

bool state = true;

void loop()
{
  uint8_t result;
    
  result=node.readHoldingRegisters(NANO_HSP(18,11), 10); // 
 
  if (result == node.ku8MBSuccess)
  {
    Serial.println("Data: ");
   // Serial.println(node.getResponseBuffer(1));
    for (uint8_t j = 0; j < 10; j++)  Serial.println(node.getResponseBuffer(j),DEC);
  
                  mySerial.flush();
                 // node.clearTransmitBuffer();
  }
  else {
    Serial.println("NO- RESPONSE");
    //mySerial.flush();
  }

  delay(3000);
  Serial.println("running...");
}

#############
#####THE Crash from the ESP8266###
Soft WDT reset

ctx: cont 
sp: 3ffef4f0 end: 3ffef820 offset: 01b0

>>>stack>>>
3ffef6a0:  feefeffe feefeffe feefeffe 00000008  
3ffef6b0:  00000008 00000003 3ffee4f4 402020ba  
3ffef6c0:  12070301 bc640a00 006f0000 014d01de  
3ffef6d0:  7f0080bc c03930ff feefefda feefeffe  
3ffef6e0:  feefeffe feefeffe feefeffe feefeffe  
3ffef6f0:  feefeffe feefeffe feefeffe feefeffe  
3ffef700:  feefeffe feefeffe feefeffe feefeffe  
3ffef710:  feefeffe feefeffe feefeffe feefeffe  
3ffef720:  feefeffe 00000004 3ffef78f 40202e18  
3ffef730:  feefeffe feefeffe feefeffe 3ffee73c  
3ffef740:  00000031 0000000a 3ffee73c 40202721  
3ffef750:  3ffef78b feefeffe feefeffe feefeffe  
3ffef760:  feefeffe feefeffe 3ffef78a 402027d5  
3ffef770:  feefeffe feefeffe feefeffe 00004b00  
3ffef780:  0000001c 00000000 31fee73c 35343332  
3ffef790:  feefef00 00000001 3ffe8669 40202e18  
3ffef7a0:  00000009 00000009 3ffe8459 3ffee7f0  
3ffef7b0:  3ffee4f4 0000000a 3ffee73c 40202721  
3ffef7c0:  0000381a 000000e2 3ffee73c 40202721  
3ffef7d0:  3ffe8450 3ffee7f0 3ffee73c 3ffee7f0  
3ffef7e0:  3ffee4f4 00000000 3ffee4f4 40202260  
3ffef7f0:  00000000 00000000 00000001 40201d38  
3ffef800:  3fffdad0 00000000 3ffee7e8 40202ba4  
3ffef810:  feefeffe feefeffe 3ffee800 40100708  
<<<stack<<<

##############################

thanks for any  guidance.

sathiyakumar3, 2017-12-04:
I too have a similar problem, the esp8266 getting soft reset every time, the modbus device goes offline

icandura, 2018-01-27:
I got same problem, too. But the chip what I use is SP3485 rather than MAX3485.

icandura, 2018-02-01:
I think I found one of the reasons for this problem.
When my MODBUS_MASTER is working under the software serial port, it resets by WDT when the slave disconnects.
But when my MODBUS master works on the hardware serial port, it works normally no matter the slave is disconnected or not.
So I think the problem lies in the software serial port.

TLS1000, 2018-05-28:
Hi, i'm having the exact same problem. I'm using MAX3485 with level shifting. Did someone find a solution?

TLS1000, 2018-05-29:
hi, I went a little deeper on the working of the software wathdog of the ESP8266.
Problem is the waiting on a response from a modbus slave (espacially during testing with no slave connected), during that time the software wathdog fires because it thinks the program hangs.

you can solve it elegantly by disabling the watchdog during modbus reads:
ESP.wdtDisable();
  result = node.readHoldingRegisters(15206,3);
  ESP.wdtEnable(1);

I have no more soft WDT resets anymore

Warlib1975, 2019-01-31:
> hi, I went a little deeper on the working of the software wathdog of the ESP8266.
> Problem is the waiting on a response from a modbus slave (espacially during testing with no slave connected), during [...]

Seems it works. Now it works more stable, haven't seen any WDT reset.