4-20ma/ModbusMaster #91: Arduino slave exemple

MarcoeSilvia, 2017-05-13:
Good morning, meanwhile congratulations to the library.
Is it possible to have an example for the slave? I searched everywhere and could not find it. Thank you very much.

Vincent0892, 2018-01-11:
Hi MarcoSivia:

Have you found the answer how to setup a slave?
I am also curious about your question.
Thank you.

vmiceli, 2018-01-15:
Here is a slave example using a different library:

`#include <modbus.h>
#include <modbusDevice.h>
#include <modbusRegBank.h>
#include <modbusSlave.h>

//Setup the brewtrollers register bank
//All of the data accumulated will be stored here
modbusDevice regBank;

//Create the modbus slave protocol handler
modbusSlave slave;

#define RS485TxEnablePin 2
#define RS485Baud 9600
#define RS485Format SERIAL_8E1

#define LED1 3
#define LED2 4

void setup()
{   
//Assign the modbus device ID.  
  regBank.setId(1);

/*
modbus registers follow the following format
00001-09999  Digital Outputs, A master device can read and write to these registers
10001-19999  Digital Inputs, A master device can only read the values from these registers
30001-39999  Analog Inputs, A master device can only read the values from these registers
40001-49999  Analog Outputs, A master device can read and write to these registers 

Analog values are 16 bit unsigned words stored with a range of 0-32767
Digital values are stored as bytes, a zero value is OFF and any nonzer value is ON

*/

//Add Digital Output registers
regBank.add(00001);
regBank.add(00002);

//Add Analog Input registers to the register bank
  regBank.add(30001);  
  regBank.add(30002);  
 

//Add Analog Output registers to the register bank
  regBank.add(40001);  
  regBank.add(40002);

  slave._device = &regBank;  

  slave.setBaud(&Serial,RS485Baud,RS485Format,RS485TxEnablePin);   
pinMode(LED1, OUTPUT);
pinMode(LED2, OUTPUT); 
}

void loop()
{
  digitalWrite(LED1, regBank.get(00001));
  digitalWrite(LED2, regBank.get(00002));
  regBank.set(30001, (word) analogRead(A0)); //from 0 - 1023
  regBank.set(30002, (word) analogRead(A1)); //from 0 - 1023
  slave.run();  
}`

yashupatel, 2018-03-31:
Hardware serial error