4-20ma/ModbusMaster #93: RS485 DE signal deactivated too early during write

gkasprow, 2017-10-14:
### ModbusMaster version
[2.0.1]

### Arduino IDE version
[1.8.5]

### Arduino Hardware
[ESP8266]

### Platform Details
[WIN10]

---

### Scenario:
[I want to communicate with MODBUS relay]

### Steps to Reproduce:
[Just use  node.writeSingleCoil(0x0000, 1); at 9600bps]

### Expected Result:
[ relay on]

### Actual Result:
[no reaction. The DE signal gets deactivated one byte too early, in the middle of CRC. The code does not take into account that UART in ESP8266 is buffered and outputs data with delay]

### Quick fix:
I simply added one line of code that re-transmits last byte;
  // transmit request
  if (_preTransmission)
  {
    _preTransmission();
  }
  for (i = 0; i < u8ModbusADUSize; i++)
  {
    _serial->write(u8ModbusADU[i]);
  }
  _serial->write(u8ModbusADU[i]);   /quick fix due to ESP8266 UART delay

vmiceli, 2018-01-14:
I just found out the hard way.... See attached the Master TX and the Slave RX. The slave always misses the last piece failing the CRC.

Can someone please elaborate on the proper fix for the ESP8266?

Thx. V.

[image]

gkasprow, 2018-01-14:
You can apply the fix above. The slave misses CRC because DE line gets deactivated too early by master.

vmiceli, 2018-01-14:
The dirty way I fixed is to add a 2ms delay in the main sketch where the postTransmission function is declared:

void postTransmission()
{
	delay(2);	//DE is pulled down too quiclky on ESP8266 and cuts off the Modbus message in the CRC
	digitalWrite(MAX485_RE_NEG, 0);
	digitalWrite(MAX485_DE, 0);
}

No delay:

[image]

With delay:

[image]

yet-another-average-joe, 2019-09-16:
See : https://github.com/4-20ma/ModbusMaster/issues/126