4-20ma/ModbusMaster #96: compatibility problem with ESP32?

TLS1000, 2018-01-10:
### ModbusMaster version
[Version of the project where you are encountering the issue]

### Arduino IDE version
[Version of Arduino IDE in your environment]

### Arduino Hardware
[Hardware information, including board and processor]

### Platform Details
[Operating system distribution and release version]

---

### Scenario:
[What you are trying to achieve and you can't?]

### Steps to Reproduce:
[If you are filing an issue what are the things we need to do in order to repro your problem? How are you using this project or any resources it includes?]

### Expected Result:
[What are you expecting to happen as the consequence of above reproduction steps?]

### Actual Result:
[What actually happens after the reproduction steps? Include the error output or a link to a gist if possible.]

---

### Feature Request

#### Narrative:

```` text
As a [role]
I want [feature]
So that [benefit]
````

#### Acceptance Criteria:

```` text
Scenario 1: Title
Given [context]
  And [some more context]...
When  [event]
Then  [outcome]
  And [another outcome]...
````

TLS1000, 2018-01-11:
I found out it's definitely a software issue. The first read after reboot works, then most reads result in E2(timeout), but sometimes there is a response from the slave device and but that results in E0 (wrong slave address) error. Please help...

TLS1000, 2018-01-11:
I found the problem. There is a problem with UART Serial(2). If you use Serial(1) it keeps working.

JacoFourie, 2018-02-24:
Hi. Can you post your code sample. Are you using a MAX485 TTL to RS485 board ?

TLS1000, 2018-02-25:
Hi, sample code: https://github.com/espressif/arduino-esp32/issues/991#issuecomment-357539051

and yes, I'm using the MAX485 board, it runs fine on CMOS levels

JacoFourie, 2018-02-25:
thanks for the info. i can see that I am sending the data out the ttl side because when I hookup a ttl to usb then I can see data. But nothing happens on the rs485 side. how did you wire it? you say it works with 3.3 volt as the board stated it is a 5v board.

TLS1000, 2018-02-25:
you need to tie DE an RE together, and define in the pre- and post transmission routines to set this pin 'high' to transmit, and a 'low' to receive. Easiest way to verify is by scope. if you son't see a pulsetrain between the A-B lines, try swapping TX and RX.

JacoFourie, 2018-02-25:
Also can you post your full code. You state you need to add a delay but is that before or after ?

JacoFourie, 2018-02-25:
Did you connect VCC to the 3.3v of the ESP32 and ground to the ground of the ESP32 on the RS485 side ?

JacoFourie, 2018-02-25:
I have a board like this one https://www.aliexpress.com/item/2PCS-MAX485-module-RS-485-module-TTL-to-RS-485-module/2055143247.html?spm=2114.search0104.3.1.178775fbdCQVtF&ws_ab_test=searchweb0_0,searchweb201602_5_10152_10151_10065_10344_10068_10342_10343_10340_10341_10084_10083_10618_10630_10307_5711211_5722315_10313_10059_10534_100031_10103_10627_10626_10624_10623_10622_5711315_10621_10620_10142,searchweb201603_25,ppcSwitch_5&algo_expid=09faa6d7-7220-46b1-bade-6fe46f4c8feb-0&algo_pvid=09faa6d7-7220-46b1-bade-6fe46f4c8feb&priceBeautifyAB=0

TLS1000, 2018-02-26:
Hi, I cannot post full code since it's a professional application. But yes, I'm using a very similar board, powered by 3,3V and ground coming from the ESP32. The delays are not so important. You just need to make sure when you're sending modbus commands, the previous ones had the time to execute. You can do this by adding delays, but there are also better ways.

TLS1000, 2018-02-26:
post you schematic and code, I'll check

TLS1000, 2018-02-26:
you're trying to read modbus information from the Serial-UART??? your modbus-uart is called 'modbusSerial', so you should try to read from that one.

you're also making it complicated by trying to decypher the modbusSerial info yourself. Try using the node.readmultipleregisters(start index, quantity) followed by node.getresponsebuffer(index)

And by using this large delay in the post-tranmission loop you're probable missing the modbus slave answer.

TLS1000, 2018-02-26:
RS485 is differential transmission so you should measure between A-B

JacoFourie, 2018-02-26:
Hi. MobusSerial is just a variable name of the UART handle of UART1. I could call it Sarial1 also. I like to keep descriptive names. I use a three UARTS on the ESP32. So giving it a name helps to make the code readable.

I do pass the modbus master the modbus serial handlel

''''
node.begin(1, ModbusSerial);
''''

Where you see that I did a printline to mobusSerial I tested to see if I could see data on the line. That is now commented out.

 I saw that the data would be scrambled if I use all 3 at the same time. I added the Flush and that will fix the UART and then all 3 work fine,

Here is a better diagram.

[image]

JacoFourie, 2018-02-26:
I dont understand this statement

"you're also making it complicated by trying to decypher the modbusSerial info yourself. Try using the node.readmultipleregisters(start index, quantity) followed by node.getresponsebuffer(index)"

Where do you see I am reading the modbus data myself? I am doing it via the Modbus Master lib.

JacoFourie, 2018-02-26:
The Checkserial stuff is my command structure that I use to send commands via the serial port. If I send a U1 it will call the function stop_vsd() that will call via modbus. U2 will start and U3 will read the register. That is on serial line that the serial monitor works on.

TLS1000, 2018-02-26:
I see. you should definitely see the pulse train on the AB lines, no use in looking further before you see a pulsetrain. If you have pulses on Tx, and RE/DE is at the right level, you should see pulses between AB....

and the node.readholdingregisters doesn't return the data, it returns a constant indicating success/failure to transmit. You should read from the responsebuffer.

I only got serial(1) working, there's a problem with serial(2), but maybe you've fixed it with that flush() command.

JacoFourie, 2018-02-26:
OK let me have a look. I have code that is already working on a Raspberry Pi that I wrote in Python. I am just re writing it to see if all will work on the ESP32. That is why I thought I had to use  

````
run_speed = node.readHoldingRegisters(0x101F , 0);
````

as this is what I did in Python using minimalmodbus

````

runspeed = vsd.read_register(0x101F, 0, functioncode=3)

````

So does readHoldingRegisters only do the function and I need to read it from a buffer somewhere ?

JacoFourie, 2018-02-26:
Did I do the wiring correct ? I do not see any pulses on the A B side. But maybe my scope does not pick it up as I have a wifi iPad scope.

JacoFourie, 2018-02-26:
OK I see pulses on the A and B side. But I have to use ground and A oe B not between A and B. So let me check the cabels from the RS485 module first.

[image]

JacoFourie, 2018-02-26:
OK I got it working. 
I had a look at my Python code again and saw the Parity was  Even.
I checked the serial setup to 

````
ModbusSerial.begin(9600 ,SERIAL_8E1, 4, 15);
````

Now I can start and stop. Still need to figure out the read but at least now I have comms. Thanks for the help.

TLS1000, 2018-02-27:
no problem, good you got it working. Did you get it working with the third UART, using serial(2)?

JacoFourie, 2018-02-27:
I am use al 3 UART now. 

````
  HardwareSerial ActuatorSerial(1);
  HardwareSerial ModbusSerial(2);

  Serial.begin(115200);
  ActuatorSerial.begin(115200, SERIAL_8N1, 16, 17);
  ModbusSerial.begin(9600 ,SERIAL_8E1, 4, 15);
````

TLS1000, 2018-04-10:
For some reason my new ESP32 doit V1 has different issues than the ones I received in the past. 
First of all there is the timing issue between the serialTXcontrol and the TX-data, this could be fixed by adding 2ms delay in the posttransmission routine. Also documented here: https://github.com/4-20ma/ModbusMaster/issues/93

The thing that bothers me the most is that ESP uart doesn't read the slave information correct: my slave responds about 25ms after the master request, by then holding the A-B positive for about 40ms before sending the dataset. The dirty fix was to hold the DE+RE high in the posttransmission routine(~50ms) untill about in the middle of this 40ms A-B positive. I don't know why the ESP32 uart doesn't read the data correct if I don't add this delay.  It doesn't work with 10ms, or with 100ms, it needs to be 50ms and that worries me :-) An arduino mega works fine with the same MAX485 driver board

JacoFourie, 2018-04-10:
Did you fix the esp32-hal-uart.c code ?. You have to comment out 

````
    //uart->dev->conf0.txfifo_rst = 1;
    //uart->dev->conf0.txfifo_rst = 0;

    //uart->dev->conf0.rxfifo_rst = 1;
    //uart->dev->conf0.rxfifo_rst = 0;

````

TLS1000, 2018-04-10:
Hi Jaco, I'll try that tomorrow. I just compared scope images on the working max485 driver on arduino mega (TTL 5V), and then the same driver on the ESP32(CMOS 3.3V). They are very different:

on the ESP:
- DI(TX) appears on the RO(RX) line during write (with RE & DE high)
- when RE+DE is low after write from master, the RO(RX) line is ~0V

on the mega (where everything works):
- RI(TX) data does not appear on the RO(RX) line during write (with RE & DE high)
- when RE+DE is low after write from master, the receiver out line is ~5V
 
I think there are different versions of the MAX485 (bought on ALIEX), my latest batch doesn't seem to like CMOS 3.3V level. I'll also try some level shifting tomorrow

TLS1000, 2018-04-11:
Hallelujah, got it working. This MAX485 driver board (the one we all use?); contrary to what I claimed earlyer, they don't all work on 3.3V although the chips appear to be the same type!!! 
So I powered it from 5V (USB voltage, Vin), driver inputs RE/DE and TX can be connected directly to the ESP as <0.8V is low and +2V is high according to MAX485 datasheet. Only on the receiver output (RO) I made resistor divider 3,4K/6,8K for not overloading the ESP's RX input.

I tried the haluart fix, but it doesn't work anymore after that, so changing nothing is better for me :-)

So this setup works...

JacoFourie, 2018-04-11:
I have ordered these 2 3.3v and 5 v ttl to RS-485 boards.

https://www.aliexpress.com/item/Stable-UART-Serial-Port-to-RS485-Converter-Function-Module-RS485-to-TTL-Module/32442485555.html

https://www.aliexpress.com/item/TTL-to-RS485-Module-RS485-Signal-Converter-3V-5-5V-Isolated-Single-Chip-Serial-Port-UART/32817054584.html

Will see how they are when I get them.

TLS1000, 2018-04-12:
interesting, let us know your feedback when you've tested them.

closed 2018-04-12 by TLS1000

MohanBailapudi, 2018-08-28:
Hi, can you please provide the working code.

MohanBailapudi, 2018-08-28:
I am getting responsetimeout error, while reading holding resisters

TLS1000, 2018-08-28:
my code is derived from the example sketches provided by the 4-20mA Modbus library. Just comment out the driver enable lines. Maybe the problem is your application that host doesn't respond or incorrect wiring? You could check with scope if signal is sent via AB-lines to the host.

MohanBailapudi, 2018-08-29:
Hi, 
Below is the code I am using, please check and let me know if doing it correct.

#include <HardwareSerial.h>
#include <ModbusMaster.h>

HardwareSerial MySerial1(1);
ModbusMaster node;
#define SSerialTxControl 7

void preTransmission()
{
  
  MySerial1.flush();
  digitalWrite(SSerialTxControl, HIGH);
  Serial.println("Transmit On");
}

void postTransmission()
{
  delayMicroseconds(120);
  digitalWrite(SSerialTxControl, LOW);
  Serial.println("Transmit Off");

}

void setup() {
      MySerial1.begin(115200, SERIAL_8N1,16,17);
    Serial.begin(115200);
    node.begin(1,MySerial1);
    //MySerial1.setDebugOutput(HIGH);
    Serial.println(MySerial1.read());
    pinMode(SSerialTxControl, OUTPUT); 
    digitalWrite(SSerialTxControl, LOW);
  
    // Callbacks allow us to configure the RS485 transceiver correctly
    node.preTransmission(preTransmission);
    node.postTransmission(postTransmission);
    
}

void loop() {
    uint8_t j, result;
    uint16_t data[1];
    result = node.readHoldingRegisters(520,1);
    Serial.println(result);
    if (result == node.ku8MBSuccess)
      {
        Serial.print("Hi");
         for (j = 0; j < 1; j++)
           {
             data[j] = node.getResponseBuffer(j);
             Serial.print(data[j]);
             
           }
      }
 }

TLS1000, 2018-08-29:
Using a max485 on an ESP32, it can be done but I don't recommend it because it's made for 5V operation and it behaves funny on 3.3V. Instead I strongly suggest to use the hardware from my last link, that one is compatible with 3.3V and very easy to use.
 About your code: I don't do the flush() command, and pretransmission and posttransmission functions can be left empty, no need to control a DE line with hardware from my latest link. the board has autodetect when to enable it's drivers. Besides that I think your code should work. Good luck

diegodaga, 2019-05-24:
Hi, i need help about that, i have a TTGO LORA32 V2.1.6 and a TTL to RS485 module same to it used Jaco, i need to write and read holding registes from a PLC and i made this code:

#include <ModbusMaster.h>
#include "SSD1306.h"

#define MAX485_DE   36
#define MAX485_RE_NEG 39

SSD1306 display(0x3c, 21, 22);

uint16_t resultMain;
uint8_t resultMain1;
int i = 250;
  
ModbusMaster node;

HardwareSerial ExtraSerial1(1);
void preTransmission()
{
  //ExtraSerial1.flush();
  digitalWrite(MAX485_RE_NEG, 1);
  digitalWrite(MAX485_DE, 1);
  //Serial.println("Transmit On");
}

void postTransmission()
{
  //delay(200);
  digitalWrite(MAX485_RE_NEG, 0);
  digitalWrite(MAX485_DE, 0);
  //Serial.println("Transmit Off");
}
void setup() {

  display.init();
  
  pinMode(MAX485_RE_NEG, OUTPUT);
  pinMode(MAX485_DE, OUTPUT);

  //init int receive mode
  digitalWrite(MAX485_RE_NEG, 0);
  digitalWrite(MAX485_DE, 0);
  
  Serial.begin(115200);
  //Modbus communication runs at 115200 baud
  ExtraSerial1.begin(9600);

  //Modbus Slave ID 1

  node.begin(1, ExtraSerial1);

  //callbacks allow us to configure the RS485 transceiver correctly
  node.preTransmission(preTransmission);
  node.postTransmission(postTransmission);

}

void loop() {

  display.clear();
  display.drawString(0,0, "resultMain1: " + String(resultMain1,HEX));
  display.display();
  
  Serial.print("resultMain1: ");
  Serial.println(resultMain1,HEX);
  resultMain1 = node.writeSingleRegister(0x01, i);
  //Serial.print("resultMain1: ");
  //Serial.println(resultMain1,HEX);

  i=i+1;

  //delay(1000);
  
  resultMain = node.readHoldingRegisters(0x00, 1);
  
  if (resultMain == node.ku8MBSuccess)
  {
    Serial.print("prueba: ");
    Serial.println(node.getResponseBuffer(0) / 100.0f);
    display.clear();
    display.drawString(0,0, "resultMain: " + String(node.getResponseBuffer(0x00)));
    display.display();
  }
  else
  {
    Serial.println("Error");
    display.clear();
    display.drawString(0,0, "No hay respuesta");
    display.display();
  }

  delay(1000);
}

the command "writeSingleRegisters" is writing fine but the answer is "0xe2" for both "resultMain" and "resultMain1",  so i can not watch any holding registers requested. Can you help me please

TLS1000, 2019-11-08:
to avoid watchdog timeouts in ESP32 when waiting for a slave response (or when no slave connected), you can disable the watchdog as described above. However this is not ideal. Also when working with multi-thread freeRTOS, the operating system should have the chance to jump to another task when waiting for slave response. 

This can be done by inserting a 1ms delay in the transmission loop, it also avoids watchdog timeouts:
Since modbus send/receive is slow, I'm using the modbus communication is a seperate freeRTOS task.

in the modbusMaster.ccp file:
```
// loop until we run out of time or bytes, or an error occurs
  u32StartTime = millis();
  while (u8BytesLeft && !u8MBStatus)
  {
    delay(1);			//WKE to avoid ESP32 watchdog timeout and reset
	if (_serial->available())
    {
#if __MODBUSMASTER_DEBUG__
      digitalWrite(__MODBUSMASTER_DEBUG_PIN_A__, true);
#endif
      u8ModbusADU[u8ModbusADUSize++] = _serial->read();
      u8BytesLeft--;
#if __MODBUSMASTER_DEBUG__
      digitalWrite(__MODBUSMASTER_DEBUG_PIN_A__, false);

```

joqr231210, 2023-07-03:
Hi guys,

Im trying to connect a ESP32 DEV KIT V1 with a MODBUS RTU Slave through this [device](https://es.aliexpress.com/item/4001028673092.html?spm=a2g0o.ppclist.product.2.772f9Ao49Ao4Eq&pdp_npi=2%40dis%21MXN%21MXN%2428%2C80%21MXN%2425%2C84%21%21%21%21%21%402103011116884169854158315e274b%2110000013574391383%21btf&_t=pvid%3A28bbe061-e207-4314-960c-27e21ad85eb2&afTraceInfo=4001028673092__pc__pcBridgePPC__xxxxxx__1688416985&gatewayAdapt=glo2esp). However, I have not received a response, even i tested it with SymplyModbusMaster, [this adapter](https://es.aliexpress.com/item/1005002669987210.html?spm=a2g0o.ppclist.product.2.397d6mTz6mTzxD&_t=pvid%3Ac6490a81-9a24-46a5-aaf9-7967f0c61425&afTraceInfo=1005002669987210__pc__pcBridgePPC__xxxxxx__1688417239&gatewayAdapt=glo2esp) and my laptop successfully:

[image]

This  is my arduino code, I am using the 4-20ma library:

"
```
#include <ModbusMaster.h>
#define MAX485_RE_NEG  2
#define MAX485_DE      4
#define RX2 16
#define TX2 17

ModbusMaster node; // Define a ModbusMaster object

void preTransmission()
{
  digitalWrite(MAX485_RE_NEG, 1); 
  digitalWrite(MAX485_DE, 1);
}

void postTransmission()
{
  digitalWrite(MAX485_RE_NEG, 0);
  digitalWrite(MAX485_DE, 0);
}

void setup()
{
  pinMode(MAX485_RE_NEG, OUTPUT);
  pinMode(MAX485_DE, OUTPUT);
  digitalWrite(MAX485_RE_NEG, 0);
  digitalWrite(MAX485_DE, 0);
  Serial.begin(9600);
  Serial1.begin(9600, SERIAL_8N1, RX2, TX2); //Define both serial channels at 9600 bps
  node.begin(1, Serial2); //Start the serial comm over node object, with Serial1, Slave ID = 1...
  node.preTransmission(preTransmission);
  node.postTransmission(postTransmission);
}

void loop()
{
  uint8_t readingResults, j, toread;
  uint16_t data1[40]; //
  toread = 4; //16 bit data registers to read... not consider zero.
  float f1,f2,f3,f4,f5,f6,f7,f8,f9,f10; //for saving the registers in float format...
  float f11,f12,f13,f14,f15,f16,f17,f18,f19,f20;
  
  readingResults = node.readHoldingRegisters(0x0036, toread); //Read address 54, toread 16-bit registers
  if (readingResults == node.ku8MBSuccess)
  {
    Serial.println("\nReadings ");
    for (j = 0; j < toread+1 ; j++)
    {
      Serial.println(">word in HEX");
      data1[j] = node.getResponseBuffer(j);
      Serial.println(data1[j], HEX);
    }
    Serial.println("\nReading ");
    memcpy(&f1,data1,4); //First block of floats
    Serial.println(f1);
    memcpy(&f2,data1+2,4);
    Serial.println(f2);
    memcpy(&f3,data1+4,4);
    Serial.println(f3);
    memcpy(&f4,data1+6,4);
    Serial.println(f4);
    memcpy(&f5,data1+8,4); 
    Serial.println(f5);
    memcpy(&f6,data1+10,4);
    Serial.println(f6);
    memcpy(&f7,data1+12,4); 
    Serial.println(f7);
    memcpy(&f8,data1+14,4);
    Serial.println(f8);
    memcpy(&f9,data1+16,4); 
    Serial.println(f9);
    memcpy(&f10,data1+18,4); 
    Serial.println(f10);

    memcpy(&f11,data1+20,4); //Second block of floats
    Serial.println(f11);
    memcpy(&f12,data1+22,4);
    Serial.println(f12);
    memcpy(&f13,data1+24,4);
    Serial.println(f13);
    memcpy(&f14,data1+26,4);
    Serial.println(f14);
    memcpy(&f15,data1+28,4); 
    Serial.println(f15);
    memcpy(&f16,data1+30,4);
    Serial.println(f16);
    memcpy(&f17,data1+32,4); 
    Serial.println(f17);
    memcpy(&f18,data1+34,4);
    Serial.println(f18);
    memcpy(&f19,data1+36,4); 
    Serial.println(f19);
    memcpy(&f20,data1+38,4); 
    Serial.println(f20);
    
  }
  else {
    Serial.print("\nConnection error: ");
    Serial.println(readingResults, HEX); 
  }
  delay(500);
} 
```
"

When wired, I get the error E0, but the slave has this address (1) and comm settings... I have read that there is an issue with ESP32 and MODBUS RTU, and I modified the Modbusmaster.cpp with a 1 ms delay in line, but then I get the E2 error... Do you recommend use the board you mentioned @TLS1000 ??? > https://www.aliexpress.com/item/Stable-UART-Serial-Port-to-RS485-Converter-Function-Module-RS485-to-TTL-Module/32442485555.html Or could you suggest something to establish communication??