4-20ma/ModbusMaster #99: Modbus communication between arduino and frequency inverter

PatrickCampos153, 2018-02-21:
### ModbusMaster version
[Version of the project where you are encountering the issue]

### Arduino IDE version
[Version of Arduino IDE in your environment]

### Arduino Hardware
[Hardware information, including board and processor]

### Platform Details
[Operating system distribution and release version]

---

### Scenario:
[What you are trying to achieve and you can't?]

### Steps to Reproduce:
[If you are filing an issue what are the things we need to do in order to repro your problem? How are you using this project or any resources it includes?]

### Expected Result:
[What are you expecting to happen as the consequence of above reproduction steps?]

### Actual Result:
[What actually happens after the reproduction steps? Include the error output or a link to a gist if possible.]

---

### Feature Request

#### Narrative:

```` text
As a [role]
I want [feature]
So that [benefit]
````

#### Acceptance Criteria:

```` text
Scenario 1: Title
Given [context]
  And [some more context]...
When  [event]
Then  [outcome]
  And [another outcome]...
````

PatrickCampos153, 2018-02-21:
209/5000
I would like to know if anyone has experience working with arduino + network modbus + inverter frequency. I have been trying to read and write parameters using this system and I was not able to succeed

bbbowden, 2018-02-21:
Are you talking about ModbusTCP or ModbusRTU?  I am using both (for different devices) and have not had any problems.

PatrickCampos153, 2018-02-21:
I am using the Modbus RTU on the RS485 media.
I now use an Arduino DUE and a ttl to RS485 conversion module. I'm having a
hard time making a communication between the Arduino and the investor.
In a better way for what you are trying to do, in case you have a knowledge
base in the area and available to you. They are very grateful.
So here's what I'm trying to do to set up a frequency inverter (cfg-11 weg
model). Use an arduino and scada oipse as a tutorial. (For example, speed,
among others) and read the inverter parameter through the supervision
system. And for communication between the elements sent, use the ModBus RTU
communication network. Did you do something like this or do you have any
knowledge?
(I apologize if something is written in a way that you do not understand,
I'm Brazilian with English in training)

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Livre
de vírus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>.
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

PatrickCampos153, 2018-02-21:
I am using Modbus RTU on RS485 media.
Now I use an Arduino DUE and a ttl to RS485 conversion module. I am having
difficulty making a communication between the Arduino and the inverter.
In a better way I will explain to you what I am trying to do if you have a
knowledge base in the area and can help me. I'll be very grateful.
So what I am trying to do is to configure the parameters of a frequency
inverter (cfg-11 weg model) using the arduino and the ellipse scada as a
supervisor (for example, configure the motor speed, among others) and I
want it the monitoring system read the inverter parameters. And for
communication between elements I am using the ModBus RTU communication
network. Did you do something like this or have you any knowledge?
(I apologize if something is written in a way that you do not understand,
I'm Brazilian with English in training)

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Livre
de vírus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>.
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

PatrickCampos153, 2018-02-26:
Do you have any knowledge of what I can do?

2018-02-21 15:47 GMT-03:00 Patrick Campos <patrickufsj@gmail.com>:

> I am using Modbus RTU on RS485 media.
> Now I use an Arduino DUE and a ttl to RS485 conversion module. I am having
> difficulty making a communication between the Arduino and the inverter.
> In a better way I will explain to you what I am trying to do if you have a
> knowledge base in the area and can help me. I'll be very grateful.
> So what I am trying to do is to configure the parameters of a frequency
> inverter (cfg-11 weg model) using the arduino and the ellipse scada as a
> supervisor (for example, configure the motor speed, among others) and I
> want it the monitoring system read the inverter parameters. And for
> communication between elements I am using the ModBus RTU communication
> network. Did you do something like this or have you any knowledge?
> (I apologize if something is written in a way that you do not understand,
> I'm Brazilian with English in training)
>
>
> <https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail> Livre
> de vírus. www.avast.com
> <https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>.
> <#m_6458695061758909096_DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>
>
>
>> Are you talking about ModbusTCP or ModbusRTU? I am using both (for
>> different devices) and have not had any problems.
>>
>>
>
>
>
> Atenciosamente,
> Patrick Campos da Silva
> Graduando em Engenharia Elétrica pela Universidade Federal de São João Del
> Rei