4-20ma/ModbusMaster #108: esp32+max3485+RS485Device

dtboy1995, 2018-06-14:
- code fragment
```c++
void postTransmission()
{ 
  //  if not write this line, the sent data is missing crc, why is this happening? and how to solve it
  //  i test write delayMicroseconds()
  delayMicroseconds(characterTime); // characterTime = 15000000 / 9600;
  digitalWrite(DE_RE, LOW);
}
```

- anthor code fragment
```c++
uint8_t result = node.readHoldingRegisters(0x104, 0x01);
Serial.println(result, HEX);
/* the serial log
0
E0
0
E0
E0
E0
0
E1
success and failure alternate, why is this happening? and how to solve it
*/

```

any help would be greatly appreciated!

amit524, 2018-06-17:
place some delay between 2 request
uint8_t result = node.readHoldingRegisters(0x104, 0x01);
Serial.println(result, HEX);
delay(50);

amit524, 2018-06-17:
delayMicroseconds(characterTime); // characterTime = 15000000 / 9600;

if you not place delay , voltage at de_re pin using meter is not identify because  of esp working very high frequency  .

As per baud rate these inter frame delay (t3.5) and inter char delay (t1.5) are as follows

1 Byte = start + 8 bits + parity + stop = 11 bits

Baud rate t 1.5 (16 bits)	t 3.5 (38 bits)
1200	13333.3	us	31666.7	us
4800	3333.3	us	7916.7	us
9600	1666.7	us	3958.3	us
19200	833.3	us	1979.2	us
38400	416.7	us	989.6	Us

dtboy1995, 2018-06-17:
@amit524 thank you very much

could you give me a sample code that suit esp32.
very thanks

closed 2018-06-26 by dtboy1995