4-20ma/ModbusMaster #116: Reading a "0x00" before the Slave Address

willian-gvs, 2019-01-02:
Hello, 

I am using the computer as Modbus Slave, ID-2, Function 03, Address 02.
The value in that address is 7.

The library is requesting correctly, and the computer is responding correctly (02 03 02 00 07 BD 86 - I checked with an oscilloscope) . 

But even so I was getting an error invalid address (0xE0).

My code is very basic.

In the setup
<code>
node.begin(2, Serial1);
</code>
In the loop
<code>  
result = node.readHoldingRegisters(0x02, 1);
</code>
<code>  
 Serial.print("Result:  ");
</code>
<code>  
 Serial.println(result, HEX);
</code>

I started to debug and changed the function ModbusMasterTransaction(uint8_t u8MBFunction)
`Serial.write(_serial->read());`
`//u8ModbusADU[u8ModbusADUSize++] = _serial->read();`
`//u8BytesLeft--;`

I am getting 
00 02 03 02 00 07 BD 86 instead of 02 03 02 00 07 BD 86.

I also notice that when there is nothing connect to the ModbusMaster, each second it is printing a 0x00.

Any tips on how I can solve this? Is anyone else having the same problem?

willian-gvs, 2019-01-03:
I am sorry, it was a hardware problem in the signal isolator.
The library is working properly.

closed 2019-01-03 by willian-gvs

turbozob, 2019-04-05:
Hello I have the same problem, how did you solve it?

willian-gvs, 2019-04-05:
I added a pull up resistor on the uart.

turbozob, 2019-04-05:
Thanks for the tip, but It did not help. This is my situation using RS485MAX module, which reports invalidSlaveID. 
Below is comm traffic from SlaveID 1

BAD message "reports 0xE0 ku8MBInvalidSlaveID"
Rx: 01 05 00 03 00 00 3D CA 00
Tx: 01 05 00 03 00 00 3D CA

GOOD "reports 0x00 ku8MBSuccess"
Rx: 01 05 00 03 00 00 3D CA
Tx: 01 05 00 03 00 00 3D CA