4-20ma/ModbusMaster #119: Error : no matching function for call to 'ModbusMaster::preTransmission(<unresolved overloaded function type>)'

apanasara, 2019-01-17:
### Scenario:
my pre/potTransmission functions & ModbusMaster Member are in a Class when I am assigning ModbusMaster::preTransmission to a class function _pretransmission  no matching function for call to 'ModbusMaster::preTransmission(<unresolved overloaded function type>)'

### Code:
```
void myClass::_preTransmission()
{
  digitalWrite(_RE, 1);
  digitalWrite(_DE, 1);
}

void myClass::begin(uint8_t sid) 
{
  _node.begin(sid, _modbusport);
   _node.preTransmission(_preTransmission); // this gives error
    //...
}
```

### Error:

> no matching function for call to 'ModbusMaster::preTransmission(<unresolved overloaded function type>)'

apanasara, 2019-01-17:
implemented [wrapper function](https://stackoverflow.com/questions/40071375/c-basics-unresolved-overloaded-function-type) but couldn't succeed

sabas1080, 2020-07-27:
@apanasara I used your variant of this library