4-20ma/ModbusMaster #125: ReadCoils not working

pruwait, 2019-08-24:
Test ModbusMaster.h and Slave simulation on PC

//read 5 coils from 1 to 6. Coils is 10101, in HEX is 15
 result = node.readCoils(0x0001,5); 
//I see in Slave LOG right ADU 01 01 01 05 ..... for RX
//I see in Slave LOG right ADU 01 01 01 (here coils in HEX) CRC  for TX

 if (result == node.ku8MBSuccess)
 {
    for (int j = 0; j < 10; j++)   Serial.print(node.getResponseBuffer(j));
     }
But I see in screen 
0 0 0 0 0 0 0 0 0 0 

What is my mistake?

bilabo438, 2020-03-31:
Hello,
i have the same problem. 
if i read Holding Registers like this: 
node.readHoldingRegisters(0,40);
it works fine.
If i read Coils like this:
node.readCoils(0,40);
i only get ceros. I have checked the Coils with a modbus programm and i should receive a 1 at adress 9.

Is there any difference for getResponseBuffer() when i use readCoils?

Thanks in advance

Rudy089, 2022-04-18:
Hi....plese check this link: https://github.com/4-20ma/ModbusMaster/issues/160#issuecomment-1100856532