4-20ma/ModbusMaster #127: [Pull Request ?] timing issues / MODBUS compliance : fix that works, with code

yet-another-average-joe, 2019-09-18:
(The solution and the code are provided below)

### ModbusMaster version
current version as of 2019/09/18

### Arduino IDE version
I don't use this thing. But gcc that comes with 1.8.9

### Arduino Hardware
slaves : Arduino Nanos running Simple Modbus Master by jecrespo (https://github.com/jecrespo/simple-modbus), Schneider-Telemecanique VFD ; master : STM32 blue Pill. had the same problem on the master side with : Mega, Due, Nano and Pro Mini.

### Platform Details
Win 10 + Visual Studio 2017 + Visual Micro - Release 1908.7.4 - 7-15th August 2019 - VS 2017-2019

---

### Scenario:

communicating reliably with all slaves, at different baudrates

### Steps to Reproduce:
just connect slaves, set baudrate at 9600 or 19200 bauds : il will not work out of the box because ModbusMaster library does not comply with the Modbus standard

### Expected Result:
ModbusMaster  should work without any hack.

### Actual Result:

timeout errors and timeout errors only ; does not works without tinkering with delays, and even doing this, it does not works for high baudrates

### Feature Request

The library should include the correct timings (T1.5 and T3.5), calculated depending on the baudrate. Doing so I was able to get MosbusMaster to communicate with the Arduino Nano slaves form 4800 baud up to 256600 bauds (Yes I know, it is more than the official max baudrate, but it worked fine).

The T3.5 delay coould be added in the _postTransmisson callback, but IMHO this is is bad, because having to do so removes one level of abstraction.

There is nothing for T1.5 (delay betwween "characters"). And this is a huge bug (the one that does that setting a baudrate greater than 19200 bauds does not work).

`delayMicroseconds(T3_5);`

must be inserted just above the postTransmission() callback function call, @line_708, just after the buffer is flushed ; this way the user can disable Tx on the MAX485 (or equivalent), and will not have to bother with delays

`delayMicroseconds(T1_5);`

*MUST* be inserted @line_752, just below the read loop. It is mandatory according to the Modbus specs. It cannot be added anywhere else except in the library sources. 
T3.5 and T1.5 computations :

```
			if (baud > 19200)
			{
				_T1_5 = 750; // 750 us
				_frameDelay = 1750; // 1750 us
			}
			else
			{
                                // _frameDalay = T3.5
				_T1_5 = 16500000 / baud; // 1 packet = always 11 bits
				_frameDelay = 38500000 / baud; // 1 packet = always 11 bits
			}

```

On my side, I slightly modded the library.

Added an init() function that also takes Baudrate, TxEnablePin, Timeout. Added variables for theses properties, made the callbacks non static (made them members so they can access the class members), made ku16MBResponseTimeout non static (static variables are the plague for those - like me - who want to run a pair of masters on the same MC !)

Doing so, I got the library to communicate with the slaves up to 256600 bauds without an error, and for hours. But I couldn't get the network to run under 4800 bauds (not sure I will investigate further as Modbus says "at leat 9600 bauds" AFAIR).

I shortly tested for higher speeds : the STM32 issues clean frames up to 4Gbps. But for real tests, I'd need to flash another STM32 with a slave firmware. Unfortunately, all my breadboards are populated at this time... But I I am pretty sure it would work.

If for some reason you don't want to do such mods (T1.5 and T3.5, at least, please :

- make timeout non static, and protected, not private !
- add a callback just below the read loop 

One last thing there is no more reason to use hungarian notation. It makes the code really difficult to read and a pain to write. I have a huge pile of code written this way, and it i now a pain. 3 or 4 years ago I was still using it. Old habits from the Windows SDK... But nowadays, editors let us know types just hovering the code with the mouse... Intellisense was on my favourite IDE in the late '90... I imagine that back in the days you also used the Win SDK...

closed 2019-10-09 by yet-another-average-joe

YagoDeleon, 2019-11-04:
Hi, Can I make the same changes for ARDUINO UNO ?, I made the same changes and didn't compile, would you have a suggestion where to put these changes?

Hantuch, 2022-02-04:
Pardon my question, but why didn't you apply the intercharacter delay inside the transmission write for-loop too?

Also another question related, the inter-frame timer is, as the modbus specs said, "at least 3.5 character times" which meant >3.5t . 
but for the inter-character timer, it said "If a silent interval of more than 1.5 character times occurs " which i infer it should be lower than 1.5 character or <1.5t . then why you use delay for that inter-character case? wouldn't a timer using if be more suitable? since delaying for that amount a time will cause as what the modbus spec said, an incomplete frame.