4-20ma/ModbusMaster #128: Unable to use readHoldingRegisters function sequentially

vignesh13404387, 2019-10-08:
I am using esp8266 as master and in my slave device, the readings are at two separate addresses.  I tried to use readHoldingRegisters function two times sequentially.   At the first time it responses with (0, HEX) but on the second time it responses (4, HEX).  

#include <ModbusMaster.h>
#define RS485_EN D11
#define RS485_RX D8
#define RS485_TX D9
ModbusMaster node;
SoftwareSerial RS485Serial(RS485_RX, RS485_TX);
void preTransmission()
{
  digitalWrite(RS485_EN, true);
}
void postTransmission()
{
  digitalWrite(RS485_EN, false);
}
void setup()
{
  Serial.begin(9600);
  RS485Serial.begin(9600);
  pinMode(RS485_EN, OUTPUT);
  digitalWrite(RS485_EN, false);
}
void loop()
{
  delay(1000);
  uint16_t result[2], value[23];
  node.preTransmission(preTransmission);
  node.postTransmission(postTransmission);
  node.begin(1, RS485Serial);
  result[0] = node.readHoldingRegisters(4096, 34);
  Serial.println(result[1], HEX); // this prints 0x00
  if (result[0] == node.ku8MBSuccess)
  {
    for (uint8_t i = 0; i <= 34; i++)
    {
      Serial.println(node.getResponseBuffer(i));
    }
  }
  node.clearResponseBuffer();
  delay(1000);
  result[1] = node.readHoldingRegisters(16384, 25);
  Serial.println(result[1], HEX);   // this prints 0x04
  if (result[1] == node.ku8MBSuccess)
  {
    for (uint8_t i = 0; i <= 25; i++)
    {
       Serial.println(node.getResponseBuffer(i));
    }
  }
  node.clearResponseBuffer();
}

YagoDeleon, 2019-11-20:
On documentation "result" equal 0x04 represents :
// static const uint8_t ku8MBSlaveDeviceFailure = 0x04
//Modbus protocol slave device failure exception.

Haminacan, 2020-02-12:
I had to add a 10 millisecond delay between requests to get it to work for the device I was using.