4-20ma/ModbusMaster #138: How can I send a data by serial communication, using an Arduino to some program that will read this data?

Pedro0596, 2020-03-27:
I never used ModBus protocol, so I'm realy confuse with all this informations. I just need to send data from arduino to a computer. It's not a big deal, but i tryied ModbusSerial and not worked. So I think in use ModbusMaster, and use arduino like a master and PC like a slave, what gonna read this data. Make sense?

IoTThinks, 2020-03-28:
You need a  TTL RS485 in your Arduino.
And USB RS485 in your PC.

If you want to send raw data between them, you don't need Modbus protocol

[image]

If you need to use RS485 sensor, you need Modbus.
Sample project for ESP32+RS485+ModbusMaster: https://github.com/IoTThinks/Misc/tree/master/ModbusMasterESP32

Pedro0596, 2020-03-28:
Right! I will try! Thanks!
The protocol is to simulate an arduilo as a PLC to test the program