4-20ma/ModbusMaster #141: Is there an asynchronous API option?

juleq, 2020-05-14:
Hi there,

I am using `readHoldingRegisters()` and `getResponseBuffer()` to read two words from a device. Unfortunately the device takes ~400ms to answer and the `readHoldingRegisters` is blocking. I would like an API that offers either a callback or a completion flag that I can poll.

An implementation of `readHoldingRegisters()` would look like this:
- Transmit the read command
- Return control to caller immediately after that instead of a blocking receive
- Regularly check for Serial.available by having the user all a `task()` function in his loop
- Call a callback on completion or set a flag that the user can check

Maybe I have missed something and that is already available? Would seem natural when the alternative is that the whole controller is dead in the water while the slave is twiddling its thumbs :).

nmaas87, 2023-12-01:
That would be really awesome, but I have also not yet found a solution for that.

juleq, 2023-12-01:
It has been a while, but I did that for my employer at the time. It was not a product and contributing back was not an option at the time. It would have been a showstopper otherwise, it was a controller for multiple PSUs and loads and it could not effort that blocking.

That said, I can still offer some comforting words :). I recall that to have been rather straightforward. So I can encourage you to give it a shot. Sorry that it's not more from me.