4-20ma/ModbusMaster #145: Retriving 0xE0 (Invalid Slave ID) Continious

nimish21, 2020-07-31:
### ModbusMaster version
[Version of the project where you are encountering the issue]

### Arduino IDE version
v1.8.7

### Arduino Hardware
STM32F0XXX

### Platform Details
Windows10 OS 

---

### Scenario:
Trying to get the data from Modbus energy meter!

### Steps to Reproduce:
I am encountering the Invalid Slave Address even the response on Modbus was [Poll->{01}{03}{00}{9C}{00}{02}{04}{25} Response->{01}{03}{04}{29}{6E}{42}{47}{E3}{20}]

### Expected Result:
Retrieve the Energy meter values

---

### Feature Request

#### Code:

```` text
#include "ModbusMaster.h"

#define MAX485_CNT     PA4

ModbusMaster node;

void preTransmission(){ digitalWrite(MAX485_CNT, 1); }
void postTransmission(){ digitalWrite(MAX485_CNT, 0); }

void setup(){
  pinMode(MAX485_CNT, OUTPUT);
  
  // Init in receive mode
  digitalWrite(MAX485_CNT, 0);

  Serial1.begin(115200);
  Serial2.begin(9600);

  
  Serial1.println("+== Initalized ==+");

  node.begin(1, Serial2);
  node.preTransmission(preTransmission);
  node.postTransmission(postTransmission);
}

void loop(){
  uint8_t result;
  result = node.readHoldingRegisters(156, 2);
  Serial1.print("0x");
  Serial1.println(result, HEX);

  if (result == node.ku8MBSuccess) {
    Serial1.print("Vbatt: ");
    Serial1.println((node.getResponseBuffer(0) + node.getResponseBuffer(1) << 16)/100.0f);
  }
  delay(1000);
}
````

sarathyplkr5, 2020-08-07:
Hi Nimish,

Are you getting the response:{01}{03}{04}{29}{6E}{42}{47}{E3}{20}], just by doing Serial1.println(result, HEX);. I am always getting an E2 (which is a time out) when I do this. I would like to see the exception code in the response. Any ideas?

v0l, 2020-09-29:
Exact same for me

v0l, 2020-09-29:
For me the first byte was always `0x00`, so i just modified the code to skip the first byte and its ok now.

maaz-shaikh, 2020-10-17:
@v0l can you post your modified code as I am also getting the same issue

bardakcioglu, 2020-12-18:
> For me the first byte was always `0x00`, so i just modified the code to skip the first byte and its ok now.
@v0l I am facing the same problem too. How can we skip the first byte, can you help us?

v0l, 2020-12-18:
@bardakcioglu something like this, https://github.com/v0l/ModbusMaster/commit/f601cca32e1c5e1ebde58306b828754204179b48

bardakcioglu, 2020-12-18:
> @bardakcioglu something like this, [v0l@f601cca](https://github.com/v0l/ModbusMaster/commit/f601cca32e1c5e1ebde58306b828754204179b48)

@v0l Thank you for your comment.
I made your changes, but the situation is still the same, I can send data to the device, this works, but I cannot see the response of this request.

17:09:21.840 ->  id: ku8MBInvalidSlaveID3
17:09:22.821 -> 
17:09:22.868 -> 0xE0

this is my serial port monitor.