4-20ma/ModbusMaster #150: wrong data received

yuguifly, 2020-11-22:
Hello, I'm trying communicating with a PLC with this library but I get a bad data and I don't know what is wrong.
the image 1 is PLC configuration, the image 2 first line is the data I want recived, the image 3 is the documentation where this data are and 
this is the code I`m using to comunicate with PLC and the response I`m getting.
Some one could help me, I'he tried a lot of things but I don't what's the problem.
Thank you.
[image]
[image]
[image]

CODE
#define MAX485_DE      3
#define MAX485_RE_NEG  2

// instantiate ModbusMaster object
ModbusMaster node;

void preTransmission()
{
  digitalWrite(MAX485_RE_NEG, 1);
  digitalWrite(MAX485_DE, 1);
}

void postTransmission()
{
  digitalWrite(MAX485_RE_NEG, 0);
  digitalWrite(MAX485_DE, 0);
}

void setup()
{
  pinMode(MAX485_RE_NEG, OUTPUT);
  pinMode(MAX485_DE, OUTPUT);
  // Init in receive mode
  digitalWrite(MAX485_RE_NEG, 0);
  digitalWrite(MAX485_DE, 0);

  Serial.begin(9200,SERIAL_8E1);

  // Modbus slave ID 1
  node.begin(49, Serial);
  // Callbacks allow us to configure the RS485 transceiver correctly
  node.preTransmission(preTransmission);
  node.postTransmission(postTransmission);
}
bool state = true;

void loop()
{
  uint8_t result;
  uint16_t data[6];
   
  // Read 16 registers starting at 0x3100)
  result = node.readHoldingRegisters(0x511D, 2);
  Serial.println("---- ENTRADA 1 ----");
  if (result == node.ku8MBSuccess)
  {
      
    data[0]=node.getResponseBuffer(0x00);
    Serial.print("Valor de REG1: ");
    Serial.println(data[0]);
    data[1]=node.getResponseBuffer(0x01);
    Serial.print("Valor de REG2: ");
    Serial.println(data[1]);
    
  }

  delay(1000);
}

Response
1Q A---- ENTRADA 1 ----
Valor de REG1q: 52429
Valor de REG2: 16649

Gio-dot, 2020-11-22:
I see 19200 in the plc serial configuration and 9200 in your code...

yuguifly, 2020-11-23:
sorry, I've put wrong in the post, I had 19200 on the test

yuguifly, 2020-11-30:
Hello I solved the problem... the plc`s documentation is wrong!!!! I have to read a position less...  the register 511D is 511C,

mridah, 2021-12-16:
Can you please tell your wiring? I'm trying to use arduino uno with Max485 but all I keep getting is 226

IoTThinks, 2021-12-16:
> Can you please tell your wiring? I'm trying to use arduino uno with Max485 but all I keep getting is 226

Try to wire TX to TX and RX to RX.

rudro101, 2022-06-26:
> Can you please tell your wiring? I'm trying to use arduino uno with Max485 but all I keep getting is 226

Did you able to solve the problem?