4-20ma/ModbusMaster #153: No Modbus slave station non-stop reset

bczhong, 2021-01-12:
BUG REPORTS - No Modbus slave station non-stop reset

### ModbusMaster version
[2.0.1]

### Arduino IDE version
[1.8.9]

### Arduino Hardware
[wemos D1R1]

### Platform Details
[VSCODE +PlatformIO]

The solution:
Add yield() after line 747 of the modbusmaster.cpp file;

SilviuMarchidanu, 2021-03-29:
I have the same problem same IDE and Version  , [wemos D1R1 (ESP8266)] just restarts after the first pooling .
I'm using the Hardware serial and have tried your solution " Add yield() after line 747 of the modbusmaster.cpp file;" but not fixed the problem .
Can you be more explicit where the " yield();  " needs to be inserted please .
Thank you .

bczhong, 2021-04-19:
[image]