4-20ma/ModbusMaster #159: Share code example read/write  multi slave ?

haidvams, 2021-06-13:
can someone help me?

besbar, 2022-04-06:
Dear all, I am also interested in using two (slave) RS485 sensors with a ESP32. Thanks a lot in advance !

JoeEarlam, 2022-07-18:
Hi, this is how I do it:

`void testRead(uint8_t addr)
{
  node.setSlaveID(addr);
  if (node.readHoldingRegisters(17, 35) == node.ku8MBSuccess)
  {
    for (uint8_t j = 0; j < 35; j++)
    {
      data[j] = node.getResponseBuffer(j);
      Serial.print(j); Serial.print(": "); Serial.println(data[j]);
    }
  }
}`

cent0, 2022-10-28:
Hello friend
We have the log for two device inverter is slave and display is master. Have read payload but this Comunication Gtil2 inverter not Hare modbus?
A this link i have shared log
https://github.com/cent0/Extend-wired-system-serial-uart-to-wifi/issues/1