4-20ma/ModbusMaster #167: Hi. Is it possible to edit communication parameters such as parity, stop bits and data length in ModbusMaster library? Thank You.

pklg, 2022-01-08:
### ModbusMaster version
[Version of the project where you are encountering the issue]

### Arduino IDE version
[Version of Arduino IDE in your environment]

### Arduino Hardware
[Hardware information, including board and processor]

### Platform Details
[Operating system distribution and release version]

---

### Scenario:
[What you are trying to achieve and you can't?]

### Steps to Reproduce:
[If you are filing an issue what are the things we need to do in order to repro your problem? How are you using this project or any resources it includes?]

### Expected Result:
[What are you expecting to happen as the consequence of above reproduction steps?]

### Actual Result:
[What actually happens after the reproduction steps? Include the error output or a link to a gist if possible.]

---

### Feature Request

#### Narrative:

```` text
As a [role]
I want [feature]
So that [benefit]
````

#### Acceptance Criteria:

```` text
Scenario 1: Title
Given [context]
  And [some more context]...
When  [event]
Then  [outcome]
  And [another outcome]...
````

noiasca, 2022-01-11:
Imho this makes no sense.
When you call .begin() you hand over a reference to your Serial object.

`void begin(uint8_t, Stream &serial);`

the library has nothing to do with parity, stopbits or data length. If you want to change this kind of parameters, just adopt the paramters of your Serial interface.

Hantuch, 2022-01-19:
Those parameters are determined on both ends of client & server serial comm. , and since modbusmaster only takes a Serial object, not create it, the change needs to be done on those side, not here, just as @noiasca said 

Although I did encounter a problem whilst using Arduino's own SoftwareSerial (not hardware mind you), which didn't have the options for parity & stop bits, in which case I found another more suitable software serial library here in Github

pklg, 2022-01-19:
Thank You. Now all is clear.