4-20ma/ModbusMaster #168: Read negative values

thalesmaoa, 2022-01-27:
Hi there, I just need help to understand if I'm facing a bug, or if this is an equipment problem.
I have a Yg889E and I've implemented the modbus communication with ESP8266. I followed [this link](https://solarduino.com/online-monitoring-for-digital-power-meter-model-yg889e-3sy-through-modbus-rtu-rs485-using-blynk-and-nodemcu/) to help me through.

However, when the multimeter shows negative values, I don't receive negative values. I tried to evaluate if it is a code problem or if this is a hardware problem. 

For example, here is part of my code:

```
union { uint32_t x; float f;} u;         // These codes required to merge 2 x 16 bits (2 register addresses into 1 float number
float z = u.f;

Serial.println("Real Power (W) per phase :");
u.x = ((unsigned long)node.getResponseBuffer(39) << 16) | node.getResponseBuffer(40); 
z = u.f; Pa = ( bitRead( node.getResponseBuffer(39), 15) ? z : z); Serial.print(Pa); Serial.print("; ");

u.x = ((unsigned long)node.getResponseBuffer(41) << 16) | node.getResponseBuffer(42); 
z = u.f; Pb = ( bitRead( node.getResponseBuffer(41), 15) ? z : z); Serial.print(Pb); Serial.print("; ");

u.x = ((unsigned long)node.getResponseBuffer(43) << 16) | node.getResponseBuffer(44); 
z = u.f; Pc = z; Serial.print(Pc); Serial.print("; ");
```

Using ModbusMaster, the bytes conversion using union is correct, however, I'm not sure if is the equipment, or if this is ModbusMaster.
In order to avoid to umount the system, I'm asking here first, maybe someone already faced similar problem with negative values.

Frtrillo, 2022-02-08:
Yes, I have same issue when reading negative number because it converts the bit that indicates positive or negative to a whole number

Frtrillo, 2022-02-12:
> But how do you know that the values are negative?

Thats up to you, I do it because I know the ranges I should read, so If its value <-10000, I do the substraction

thalesmaoa, 2022-02-12:
This a problem with the meter or with the lib?

Rambus100lica, 2022-10-03:
Hi, 
I have same problem. The case is that sign is held in third register and I don't know how to read three registers and make a value from them

thalesmaoa, 2023-04-24:
This repo is old. Does anyone suggest a new one?