4-20ma/ModbusMaster #177: Reading 32 bits number from two registers

Fluxanode, 2022-08-18:
I need to get a single value from a holding or input reg that is in two reg addresses.  
I want to read 8006 (8007) as the date then read 8006 (time) to store in two separate vars.

 
Floating Point (FP) 4bytes 2 registers

Holding Reg
REG    DES          Datatype
8006  Real Date  FP
8008 Real Time  FP

Can you give me an example how this is coded using your lib?

fuzonmedia, 2022-11-28:
@Fluxanode 
Do you able to solve this ? I need to know the same

Fluxanode, 2022-11-28:
I have not been working on this lately and never received an answer.  Sorry.  It would be nice if someone bothered to answer...

phaelbotelho, 2022-11-28:
So, this it's not a issue to be solved, but follow the  solution.
As the data type os the variable, you can just copy the value to a variable able to read as 2 ways. Like the code below:
`union floatto16bits{
float value_32;
uint16_t value_16[2];
};`
or
`union uint32_tto16bits{
uint32_t value_32;
uint16_t value_16[2];
};`
This type of variable, alloc the same size of memory and allow you to read as you need.

When write the two 16 bis value use this name:
Uint32_tto16bits.value_16[0]
Uint32_tto16bits.value_16[1]

When read as a 32 type use this:
Uint32_tto16bits.value_32

mftadvantage, 2022-12-23:
Olá alguem pode me ajudar, estou tentando ler dados de um medidor e recebo o padrão (IEEE 32-bit floating point). Não consigo de maneira alguma converter esse padrão em decimal. no exemplo do fabricante ele fala que 0x00007042 (IEEE 32-bit floating point) é = 60 em decimal. Mas não consigo entender nem criar um programa que faça essa conversão.

fuzonmedia, 2022-12-24:
@mftadvantage Email me your issue `niladridey933@gmail.com` I can help out