4-20ma/ModbusMaster #199: Calculation error due to using uint16_t = 65535? 66.5kW = 0.5kW!

ilya7zz, 2024-11-22:
When the active power (W) and accumulated power (Wh) measurements are exceeded, the buffer overflows.
For example:
65.535 kWh is the limit after which it starts counting from zero.
i.e. 65.537 kWh = 0.001 kWh
6553.5 W is the limit after which it starts counting from zero.
6553.7 W = 0.1 W

```
bool PR_alarm;
uint8_t result;
float U_PR, I_PR, P_PR, PPR, PR_F, PR_PF; 
void PZEM() {
  result = node.readInputRegisters(0x0000, 10);
  if (result == node.ku8MBSuccess) {
  U_PR = (node.getResponseBuffer(0x00)/10.0f); // Voltage
  I_PR = (node.getResponseBuffer(0x01)/1000.000f); // Current
  P_PR = (node.getResponseBuffer(0x03)/10.0f); // Power
  PPR = (node.getResponseBuffer(0x05)/1000.0f); // Power consumption max:9999.99
  PR_F = (node.getResponseBuffer(0x07)/10.0f); // Frequency 45 – 65 Hz
  PR_PF = (node.getResponseBuffer(0x08)/100.0f); // Power Factor
  PR_alarm = (node.getResponseBuffer(0x09)); // Active power threshold
  sendSerial();
}

void sendSerial() {
  Serial.print("U_PR: " + String(U_PR,1) + " V,  ");
  Serial.print("I_PR: " + String(I_PR,3) + " A,  ");
  Serial.print("P_PR: " + String(P_PR,1) + " W,  ");
  Serial.print("PPR: " + String(PPR,3) + " kW*h,  ");
  Serial.print("PR_F: " + String(PR_F,1) + " Hz,  ");
  Serial.print("PR_PF: " + String(PR_PF) + " cos φ,  ");    // Power factor (cos φ)
  Serial.println("PR_alarm: " + String(PR_alarm,0)); // Active power threshold
}

```
I really liked this library, I ask you to improve it!
Or please advise me how to fix this error ;)

ilya7zz, 2024-11-22:
Also, there is an error during compilation:
```
sketch/src/ModbusMaster.cpp: In member function 'uint8_t ModbusMaster::requestFrom(uint16_t, uint16_t)':
sketch/src/ModbusMaster.cpp:94:33: warning: 'read' is used uninitialized in this function [-Wuninitialized]
   _u8ResponseBufferLength = read;
                                 ^
```

Fluxanode, 2024-12-03:
I'm seeing the same "read" warning and started a new issue.  I think this lib is not maintained/updated.

ilya7zz, 2024-12-04:
I also thought that the project was abandoned. That's why I found the [PZEM-004T-v30](https://github.com/mandulaj/PZEM-004T-v30) library and switched to it. It turned out to be more convenient to work with it and the answers are accurate. However, there are errors there too :)
I [wrote](https://github.com/mandulaj/PZEM-004T-v30/issues/133) to the author, I hope for his answer.

Fluxanode, 2024-12-04:
I use it for reading industrial sensors typically flow meters.  It works but there are a lot of issues opened and just wonder what happened to the author?

HenniePeters, 2025-01-12:
Are you sure that your power meter can handle more than 65535W?
I use the ModbusMaster library for an Entes powermeter and I need to combine two 16 bit words to create an uint32_t.

uint32_t i = (uint32_t)node.getResponseBuffer( 0 ) << 16;
i += node.getResponseBuffer( 1 );
float f = (float)i / 10.0;

ilya7zz, 2025-01-14:
> Are you sure that your power meter can handle more than 65535W?
I'm sure) When switching to another library, everything works fine.
[image]
In the previous message I reported which library I switched to.

HenniePeters, 2025-01-14:
I didn't notice the gaps in the used registers before, in your code:
    I_PR = (node.getResponseBuffer(0x01)/1000.000f); // Current
    P_PR = (node.getResponseBuffer(0x03)/10.0f); // Power
    PPR = (node.getResponseBuffer(0x05)/1000.0f); // Power consumption max:9999.99
In your code the locations 2 and 4 are not used. I expect that location 2 contains the high value of P_PR and location 4 the high value of PPR.
so:
    P_PR = ( ( (uint32_t) node.getResponseBuffer(0x02) ) << 16 ) + node.getResponseBuffer(0x03) ) / 10.0f; 
    PPR =  ( ( (uint32_t) node.getResponseBuffer(0x04) ) << 16 ) + node.getResponseBuffer(0x05) ) / 1000.0f;