4-20ma/ModbusMaster #200: Compile issue warning in modbusmaster.cpp

Fluxanode, 2024-12-03:
Why am i getting this warning and is it something you need to fix?

d:\User\Arduino\libraries\ModbusMaster\src\ModbusMaster.cpp: In member function 'uint8_t ModbusMaster::requestFrom(uint16_t, uint16_t)':
d:\User\Arduino\libraries\ModbusMaster\src\ModbusMaster.cpp:94:27: warning: 'read' is used uninitialized [-Wuninitialized]
   94 |   _u8ResponseBufferLength = read;
        |   ~~~~~~~~~~~~~~~~~~^~~~~~
**Do you even maintain this library?**

HenniePeters, 2025-01-12:
Do you need that function?
I commented the whole function requestFrom and the warning is gone :-)
I don't like red texts while compiling so this seems the best solution.

Fluxanode, 2025-01-13:
And it still works?  Which function did you comment out?

Fluxanode, 2025-01-13:
Exactly what did you comment out?  I tried it and got more errors and wouldn't compile.

Fluxanode, 2025-01-13:
ok I figured that one out.  
what would you do about this one?
d:\ChuckW\Arduino\libraries\ModbusMaster\src/util/crc16.h:71:17: warning: 'uint16_t crc16_update(uint16_t, uint8_t)' defined but not used [-Wunused-function]
   71 | static uint16_t crc16_update(uint16_t crc, uint8_t a)
      |                          ^~~~~~~~~~~~
I tried commenting it out and got more errors.

HenniePeters, 2025-01-13:
I commented out: requestFrom  It was not used.
the warning from crc16 can be solved by moving the include file from the .h to the .cpp file

HenniePeters, 2025-01-13:
Remove it from ModbusMaster.h and add it to ModbusMaster.cpp