4-20ma/ModbusMaster #201: Can you help me with a case when you are free

herofence, 2025-01-01:
I want to use ESP8266 to connect PLC, use ESP8266 as the modbus master station, read multiple coils and multiple holding registers of PLC and write them into holding registers, and then output them to the WEB page for control. Can you give me a case study of this. Now I have a problem that I don't know how to publish the PLC data to the serial port and web page after reading it.

void readCoils() {
node.readCoils(0x0001,7); //Read the status of 7 coils from Q0.0 to Q0.6

}

void readHoldingRegisters() {
  node.readHoldingRegisters(0x9d3b,5); //Read VW500-VW508, inverter status
}

void writeSingleRegister() { 
  uint16_t value=0;
  node.writeSingleRegister(0x9d40,value); //Write VW510
}

miteshrabadiaaru, 2025-01-03:
This code example contains Modbus communication functions used to interact with remote devices via the Modbus protocol. Below are the details of these functions and their purposes. I have also made some changes that may be helpful to you. When you have a chance, please review the documentation and code example files, and feel free to reach out if you have any questions or issues
```cpp
/*
This function code is used to read from 1 to 2000 contiguous status of 
coils in a remote device. The request specifies the starting address, 
i.e. the address of the first coil specified, and the number of coils. 
Coils are addressed starting at zero.

The coils in the response buffer are packed as one coil per bit of the 
data field. Status is indicated as 1=ON and 0=OFF. The LSB of the first 
data word contains the output addressed in the query. The other coils 
follow toward the high order end of this word and from low order to high 
order in subsequent words.

If the returned quantity is not a multiple of sixteen, the remaining 
bits in the final data word will be padded with zeros (toward the high 
order end of the word).
*/

void readCoil() {
  uint16_t startingAddress = 0x0001;  // Starting address (e.g., coil address 0x0001)
  uint16_t quantity = 7;              // Number of coils to read

  // Send the read coils command
  uint8_t result = node.readCoils(startingAddress, quantity);

  // Check if the read was successful
  if (result == node.ku8MBSuccess) {
    Serial.println("Successfully read coils.");
    for (uint8_t i = 0; i < quantity; i++) {
      Serial.print("Coil ");
      Serial.print(i);
      Serial.print(": ");
      Serial.println(node.getResponseBuffer(i) ? "ON" : "OFF");
    }
  } else {
    Serial.print("Failed to read coils, error: ");
    Serial.println(result, HEX);
  }
}

/*
This function code is used to read the contents of a contiguous block of 
holding registers in a remote device. The request specifies the starting 
register address and the number of registers. Registers are addressed 
starting at zero.

The register data in the response buffer is packed as one word per 
register.
*/

void readHoldingRegister() {
  int16_t result = node.readHoldingRegisters(0x9d3b, 5);  //Read VW500-VW508, inverter status
  if (result == node.ku8MBSuccess) {
    // Holding resistors values
    int16_t data0 = (int16_t)node.getResponseBuffer(0);
    int16_t data1 = (int16_t)node.getResponseBuffer(1);
    int16_t data2 = (int16_t)node.getResponseBuffer(2);
    int16_t data3 = (int16_t)node.getResponseBuffer(3);
    int16_t data4 = (int16_t)node.getResponseBuffer(4);
  }
}

/*
This function code is used to write a single holding register in a 
remote device. The request specifies the address of the register to be 
written. Registers are addressed starting at zero.
@param u16WriteAddress address of the holding register (0x0000..0xFFFF)
@param u16WriteValue value to be written to holding register (0x0000..0xFFFF)
@return 0 on success; exception number on failure
*/

void writeSingleRegisters() {
  uint16_t value = 1234;  // The value you want to write to the register (replace with actual value)

  uint8_t result = node.writeSingleRegister(0x9d40, value);  // Write value to register 0x9d40

  // Check if the operation was successful
  if (result == node.ku8MBSuccess) {
    Serial.println("Successfully wrote to register.");
  } else {
    Serial.print("Failed to write to register, error: ");
    Serial.println(result, HEX);  // Print error code in hexadecimal
  }
}

herofence, 2025-01-04:
Thank you for your help, which has benefited me greatly. Thank you again.

closed 2025-01-04 by herofence