4-20ma/ModbusMaster #202: Writing to modbus function 05

Fluxanode, 2025-03-20:
anyone know how to write to modbus function register 05 using this lib?  Maybe someone?

miteshrabadiaaru, 2025-03-27:
```cpp
/*
This function code is used to write a single holding register in a 
remote device. The request specifies the address of the register to be 
written. Registers are addressed starting at zero.
@param u16WriteAddress address of the holding register (0x0000..0xFFFF)
@param u16WriteValue value to be written to holding register (0x0000..0xFFFF)
@return 0 on success; exception number on failure
*/

void writeSingleRegisters() {
  uint16_t value = 1234;  // The value you want to write to the register (replace with actual value)

  uint8_t result = node.writeSingleRegister(0x9d40, value);  // Write value to register 0x9d40

  // Check if the operation was successful
  if (result == node.ku8MBSuccess) {
    Serial.println("Successfully wrote to register.");
  } else {
    Serial.print("Failed to write to register, error: ");
    Serial.println(result, HEX);  // Print error code in hexadecimal
  }
}
````
For more information, please refer to the examples.

Fluxanode, 2025-03-27:
I figured it out, i had to write to a coil.